robotology/robot-testing-framework #4: Move test name from main configuration file to individual test configuratio files

traversaro, 2015-01-27:
In a nutshell, I propose we move the test name from the main configuration file, as in: 
```
[REFERENCES]
user "alessandro scalzo"
comment "this is a wild bunch of tests"
outfile iCubTestReport.xml

[TESTS]
TestCamera right_camera.ini
TestCamera left_camera.ini
TestMotors test_left_arm.ini
TestMotors test_right_arm.ini
TestFTSensors test_ft_left_arm.ini
TestFTSensors test_ft_right_arm.ini
TestFTSensors test_ft_left_leg.ini
TestFTSensors test_ft_right_leg.ini
TestFTSensors test_ft_left_foot.ini
TestFTSensors test_ft_right_foot.ini
```
to the test configuration file, so we will have:
```
[REFERENCES]
user "alessandro scalzo"
comment "this is a wild bunch of tests"
outfile iCubTestReport.xml

[TESTS]
right_camera.ini
left_camera.ini
test_left_arm.ini
test_right_arm.ini
test_ft_left_arm.ini
test_ft_right_arm.ini
test_ft_left_leg.ini
test_ft_right_leg.ini
test_ft_left_foot.ini
test_ft_right_foot.ini
```
This will permit to reuse more easily the same .ini file in multiple main configuration files, and is more consistent with the device use (test and device share a lot of commonality in my opinion).

lornat75, 2015-01-27:
Drawback: it requires the code that execute tests to actually open and parse individual ini files.

traversaro, 2015-01-27:
The main is already opening the .ini files, for getting the yarp::os::Property object to pass to the test.

lornat75, 2015-01-27:
but is it parsing them?

traversaro, 2015-01-27:
No, but I don't think it should increase the complexity a lot. 
Just because some lines of code are worth 1000 words, check this commit : https://github.com/robotology/robot-testing/commit/092dbd656a9dc238e25ec508ee02cc6ec5dcee0f (if this is ok I can also migrate the configuration files).

traversaro, 2015-03-21:
The new xml "suit" structure has made this issue obsolete.

closed 2015-03-21 by traversaro