robotology/robot-testing-framework #6: Robot specific parameters or tests

lornat75, 2015-01-27:
Different robots either real or simulated (ODE or gazebo) have different functionality. Even in case in which functionalities are the same, tests may need different parameters (thresholds, speeds etc).

https://github.com/robotology/robot-testing/pull/5 addresses how to propagate variables from the main configuration file to individual test files. This is for sure useful to propagate the name of the robot as a prefix for port names.

At the moment using the environment functionality (https://github.com/robotology/robot-testing/pull/5) we can use different configuration files (test-ode.ini/test-gazebo.ini)

I wonder if in the end we will need to build different directories/contexts, i.e.:

app/iCubSim --> test ode simulator
app/icub --> test real robot
app/icubGazeboSim --> gazebo simulator

traversaro, 2015-01-27:
I'll keep the simplest solution (different configuration files) as long as it is suitable.

traversaro, 2015-01-27:
Probably we will need different directories for totally different robots (coman, double_pendulum ).

traversaro, 2017-10-30:
Not relevant anymore to this repo, closing.

closed 2017-10-30 by traversaro