robotology/robot-testing-framework #8: [TestFTSensors] add use of the sensor through the AnalogSensorClient

traversaro, 2015-01-29:
Currently in TestFTSensors we are reading just the sensor port, we should also add the test of use the AnalogSensorClient to read its value, check its state and the timestamps. Reading the raw port should however remain because it is how it is used in a lot of code. 
 
cc @barbalberto any documentation on how to properly use AnalogSensorClient ?

barbalberto, 2015-02-02:
Mandatory param:

local:      full name of the local port the client will open (the data port, without the /rpc)
remote:  fulll name of the remote port to attach to  (the data port, without the /rpc)

Optional:
carrier:    type of carrier like udp, tcp and so on ...

I'll add it in the code

barbalberto, 2015-02-02:
They are already in the yarp branch under testing on the red iCub, I just added the carrier that was missing

traversaro, 2015-02-17:
Another interesting use case for ft sensor testing on the real robot is to explore all the robot workspace and check that the FT sensors is not overflowing.

traversaro, 2015-02-23:
Another test to add to the test suite  with real robots sensors is that their measure should be always different from 0.0 0.0  0.0 0.0 0.0 0.0 (cc @jeljaik )

randaz81, 2015-03-25:
in the new framework I can't find this test anymore (also testMotors disappeared).

traversaro, 2016-12-05:
The actual tests have been moved to https://github.com/robotology/icub-tests, any new issue on them can be opened there.

closed 2016-12-05 by traversaro