robotology/robot-testing-framework #12: Support dynamically loaded test

traversaro, 2015-02-05:
For now robot-testing supports just test that are hard coded (and directly linked) into the main application `robot-testing`. 

This is undesirable for the following reasons:
  * the users may want to test an own library, without linking it directly to `robot-testing`.
  * the user have to modify the main cpp file for each test that they had, that is a bit awkward . 

yarp has a nice feature of allowing dynamically loaded device drivers. I guess we can borrow most of this yarp device drivers infrastructure/logic, in a way that `robot-testing` will play a role similar to what `yarpdev` is doing for devices.
In a nutshell, the ideal scenario is the one where the CI jobs will simply call `robot-testing testListSimulation.ini` .  
Similarly a human user that want to test a real robot will call `robot-testing testListRobot.ini`.

cc @lornat75 @apaikan

lornat75, 2015-02-05:
@drdanz maybe you can help here?

apaikan, 2015-02-05:
I am coming up with a more generic and extendable architecture where robot tests can be added independently using plugins written either in dll/so or Lua/python scripts. It also has Fixture manager to setup the test suit (e.g., running siumulator, robot interface, ...) before running the tests.  many customized assertion will be provided to properly report tests/fixture or code failures and errors. In general, it can be used to develop tests in different levels: code, components, simulator, robot.  Many useful and unique features such as web-interface for test runner/monitor , XML/HTML/TEXT outputter will be provided. 

The implementation will be completely robot/middleware and OS independent and each middleware--dependent part can be added to the testing framework as separate plugin. Our current existing tests can be easily added to the new framework. 

Just wait a bit more.

lornat75, 2015-02-05:
Not sure if I have to be scared or excited...

traversaro, 2015-03-25:
I guess this issue is one of the main things solved by the RTF framework, so we can close this issue.

closed 2015-03-25 by traversaro