robotology/robot-testing-framework #15: please do not use std::cout, std::cerr, fprintf etc

randaz81, 2015-03-24:
...for logging purposes please use yInfo(), yWarning(), yError(), including <yarp/os/LogStream.h> and <yarp/os/Log.h> as appropriate.

traversaro, 2015-03-25:
I guess this can be accomplished by properly extending the `TestResultCollector` object, but @apaikan can provide more information for sure.

apaikan, 2015-03-26:
@randaz81  RTF by default does not use YARP. In general, to report the tests result at run-time, a user-defined class can be implemented which is inherited from `RTF::TestListener` (http://robotology.github.io/robot-testing/documentation/classRTF_1_1TestListener.html). 

Alternatively results can be outputted (after finishing test's run) in user-defined format by extracting the results from `RTF::TestResultCollector` as @traversaro  suggested (see `RTF::TextOutputter`). However we should think if they are needed to become separate plugins from the `testrunner` or  making RTF to be optionally compiled with YARP.

traversaro, 2015-03-27:
Just for the sake of brainstorming: while including some (optional) stuff depending on YARP in RTF could make sense, for example it will for sure simplify deployment outside of superbuilds.

However, perhaps for some user/framework specific customizations, such as  extending `RTF::TestListner` for using framework specific logging functions it could make sense to let the user add specific Callbacks to the `RTF::TestListner`, that can be configured for example by the `YarpTestCase` object. This would permit the user to customize some behavior (as in `rFSM` we do with "hooks") without having an additional plugin type.

apaikan, 2015-05-18:
RTF has a web reporter (See http://robotology.github.io/robot-testing/documentation/testrunner.html#weblistener) to remotely monitor the result of the tests via a web browser. 
So I close the issue and If we still need an integration with YARP logger, we can open a new "enhancement" issue!

closed 2015-05-18 by apaikan