robotology/robot-testing-framework #16: Approximal equality helpers

traversaro, 2015-03-25:
We need some helpers for testing approximate equality of floating point values.

An important feature to have is to print the values when the comparison fails, for easy debugging of failing tests (as in gtest `EXPECT_NEAR`, `ASSERT_NEAR`, https://code.google.com/p/googletest/wiki/AdvancedGui ),

arocchi, 2015-03-25:
I think the right way to have gtest-like functionality is to use gtest :)

traversaro, 2015-03-25:
The main problems for using directly GTest, beside the problems of dependency handling (that are not trivial, especially on windows) are:

* GTest assert functions is that they don't play well with existing log systems. See for example how ugly is this test output in Gazebo, where gtest output does not integrated with the native Gazebo logger : https://bitbucket.org/osrf/gazebo/issue/1197/dartjoint-getforcetorque-returns-wrong . This will be a problem for example if we use RTF to generate XML reports from remote users, and we want all the test output in the XML reports in a clean way, or we want this output to use to use middleware specific logging frameworks. 

* The GTest approximal equality function do not cover testing Vectors or Matrices. 

* The plan that @apaikan has is to support test cases also writting in scriptigin languages, and I guess that for doing that we will have to expose bindings of the logging/assert functions, that are not available in GTest. 

In the end, given that this approximal equality helpers can be written in probably 10 lines of code, we have to consider that the "Not Invented Here Syndrome" is a problem, but another risk is to always try to rely on external dependencies, when perhaps the simplest and more manteinable way of doing things is an ad-hoc solution.

apaikan, 2015-03-27:
Agree with @traversaro . We can provide more basic checking and assertion macros/function. They are pretty easy to  implement (some useful cppuint macros can be found here http://cppunit.sourceforge.net/doc/cvs/group___assertions.html) . 

* Notice that more complex and framework-dependent assertions should go to the `icub-tests` or equivalent tests repository.

apaikan, 2015-10-19:
A new class (`RTF::YARP::YarpTestAsserter`) has been added within the `RTF_yarp_utility` library which aggregates some of the auxiliary  checker functions (e.g., `YarpTestAsserter::isApproxEqual`).  
We can collect more helper functions as we need in future.

closed 2015-10-19 by apaikan