robotology/robot-testing-framework #20: headers install directory

traversaro, 2015-04-29:
All the RTF headers are installed in the `include`  directory, so they are used as:

~~~
#include <TestCase.h>
#include <TestAssert.h>
#include <Arguments.h>
~~~

Could it make sense to move them in `include/rtf` (or `include/RTF`), so we will have:
~~~
#include <rtf/TestCase.h>
#include <rtf/TestAssert.h>
#include <rtf/Arguments.h>
~~~

cc @apaikan

apaikan, 2015-05-08:
@traversaro well noticed! 
agreed!

traversaro, 2015-05-08:
I will probably do it in the weekend, so Monday morning @randaz81 and @ale-git can simply update `robot-testing` and `icub-tests` and get everything working.

traversaro, 2015-05-12:
Done in https://github.com/robotology/robot-testing/issues/22 .

closed 2015-05-12 by traversaro