robotology/robot-testing-framework #23: CTest integration

traversaro, 2015-05-11:
To integrate nicely with the traditional CMake/CTest  [`add_test`](http://www.cmake.org/cmake/help/v3.0/command/add_test.html) command, we need that the `testrunner` returns a number different from `0` on failure. 
For this are necessary a bunch of modifications: 
  * [ ] For reporting failed test, the `return 0` in https://github.com/robotology/robot-testing/blob/master/src/testrunner/src/main.cpp#L232 should be substituted with `return collector.failedCount();` 
  * [ ] For loading errors (i.e. failure of a `load*` method) I guess it make sense to simply return 1.

traversaro, 2015-05-11:
cc @apaikan

traversaro, 2015-05-12:
Fixed in https://github.com/robotology/robot-testing/commit/35678bdcda725802bb11ce6ca98eb10a814a6f6f . I used `cstdlib` `EXIT_FAILURE` and `EXIT_SUCCESS`.

closed 2015-05-12 by traversaro

apaikan, 2015-05-12:
@traversaro  cool!