robotology/robot-testing-framework #26: Error of type **stack smashing detected*** when using yarpmanager fixture plugin

traversaro, 2015-05-14:
I had a fatal error of this type:

~~~
pegua@pareto:~/src/icub-tests/suits$ testrunner --suit basics-icubSim.xml --verbose
Loading basics-icubSim.xml
Staring test runner.
Test suit Basic Tests Suite started...
[INFO]  (Basic Tests Suite) reports: yarpmanager is setuping the fixture...
Returning true from yarpManager setup*** stack smashing detected ***: testrunner terminated
Aborted (core dumped)
~~~

This is probably caused by some wrong parameters passed to the yarpmanager fixture plugin. I have disabled the GCC stack protection compiling `RTF` with the `-fno-stack-protector`, and I still get some error because it is not able to find the configuration files.

traversaro, 2015-05-14:
System: Ubuntu 14.04.02 .

traversaro, 2015-05-15:
Unfortunately I was able to replicate on Travis (i.e. Ubuntu 12.04), see this job:
https://travis-ci.org/robotology/gazebo-yarp-plugins/jobs/62670916 .
We can hardcode the `-fno-stack-protector` option in the CMake, but I guess something really fishy that I do not understand is going on..

traversaro, 2015-05-15:
Stack trace from Travis:
~~~bash
$ testrunner --verbose --suit ../../icub-tests/suits/basics-icubGazeboSim.xml
Loading ../../icub-tests/suits/basics-icubGazeboSim.xml
Staring test runner.
Test suit Basic Tests Suite started...
[INFO]  (Basic Tests Suite) reports: yarpmanager is setuping the fixture...
*** stack smashing detected ***: testrunner terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x37)[0x7f1556ce1007]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x0)[0x7f1556ce0fd0]
/usr/local/lib/yarpmanager.so(_ZN14YarpFixManager5setupEiPPc+0x1961)[0x7f155641c3bd]
/usr/local/lib/libRTF.so(_ZN3RTF14FixtureManager5setupEv+0xaf)[0x7f15578ec945]
/usr/local/lib/libRTF.so(_ZN3RTF8TestSuit5setupEv+0x2d)[0x7f15578f3073]
/usr/local/lib/libRTF.so(_ZN3RTF8TestSuit3runERNS_10TestResultE+0xa8)[0x7f15578f315e]
/usr/local/lib/libRTF.so(_ZN3RTF10TestRunner3runERNS_10TestResultE+0x82)[0x7f15578f18d4]
testrunner(main+0xcb5)[0x41d99c]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xed)[0x7f1556bf976d]
testrunner[0x414c79]
======= Memory map: ========
00400000-0042e000 r-xp 00000000 fd:02 10494391                           /usr/local/bin/testrunner
0062d000-0062e000 r--p 0002d000 fd:02 10494391                           /usr/local/bin/testrunner
0062e000-0062f000 rw-p 0002e000 fd:02 10494391                           /usr/local/bin/testrunner
00a48000-00b94000 rw-p 00000000 00:00 0                                  [heap]
7f153c000000-7f153c022000 rw-p 00000000 00:00 0 
7f153c022000-7f1540000000 ---p 00000000 00:00 0 
7f1544000000-7f1544022000 rw-p 00000000 00:00 0 
7f1544022000-7f1548000000 ---p 00000000 00:00 0 
7f1548000000-7f1548022000 rw-p 00000000 00:00 0 
7f1548022000-7f154c000000 ---p 00000000 00:00 0 
7f154c000000-7f154c022000 rw-p 00000000 00:00 0 
7f154c022000-7f1550000000 ---p 00000000 00:00 0 
7f1551c44000-7f1551c45000 ---p 00000000 00:00 0 
7f1551c45000-7f1552645000 rw-p 00000000 00:00 0 
7f1552645000-7f1552646000 ---p 00000000 00:00 0 
7f1552646000-7f1553046000 rw-p 00000000 00:00 0
7f1553046000-7f1553047000 ---p 00000000 00:00 0 
7f1553047000-7f1553a47000 rw-p 00000000 00:00 0 
7f1553a47000-7f1553a48000 ---p 00000000 00:00 0 
7f1553a48000-7f1554448000 rw-p 00000000 00:00 0 
7f1554448000-7f155445d000 r-xp 00000000 fd:02 5249184                    /home/travis/build/robotology/icub-tests/build/plugins/FtSensorTest.so
7f155445d000-7f155465d000 ---p 00015000 fd:02 5249184                    /home/travis/build/robotology/icub-tests/build/plugins/FtSensorTest.so
7f155465d000-7f155465e000 r--p 00015000 fd:02 5249184                    /home/travis/build/robotology/icub-tests/build/plugins/FtSensorTest.so
7f155465e000-7f155465f000 rw-p 00016000 fd:02 5249184                    /home/travis/build/robotology/icub-tests/build/plugins/FtSensorTest.so
7f1554660000-7f1554682000 r-xp 00000000 fd:02 9577973                    /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f1554682000-7f1554882000 ---p 00022000 fd:02 9577973                    /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f1554882000-7f1554886000 r--p 00022000 fd:02 9577973                    /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f1554886000-7f1554887000 rw-p 00026000 fd:02 9577973                    /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f1554888000-7f15548cc000 r-xp 00000000 fd:02 13512584                   /usr/lib/libgslcblas.so.0.0.0
7f15548cc000-7f1554acb000 ---p 00044000 fd:02 13512584                   /usr/lib/libgslcblas.so.0.0.0
7f1554acb000-7f1554acc000 r--p 00043000 fd:02 13512584                   /usr/lib/libgslcblas.so.0.0.0
7f1554acc000-7f1554acd000 rw-p 00044000 fd:02 13512584                   /usr/lib/libgslcblas.so.0.0.0
7f1554ad0000-7f1554cfb000 r-xp 00000000 fd:02 13512583                   /usr/lib/libgsl.so.0.16.0
7f1554cfb000-7f1554efa000 ---p 0022b000 fd:02 13512583                   /usr/lib/libgsl.so.0.16.0
7f1554efa000-7f1554efd000 r--p 0022a000 fd:02 13512583                   /usr/lib/libgsl.so.0.16.0
7f1554efd000-7f1554f0b000 rw-p 0022d000 fd:02 13512583                   /usr/lib/libgsl.so.0.16.0
7f1554f0b000-7f1554f0c000 rw-p 00000000 00:00 0 
7f1554f10000-7f1554f49000 r-xp 00000000 fd:02 9577957                    /lib/x86_64-linux-gnu/libreadline.so.6.2
7f1554f49000-7f1555149000 ---p 00039000 fd:02 9577957                    /lib/x86_64-linux-gnu/libreadline.so.6.2
7f1555149000-7f155514b000 r--p 00039000 fd:02 9577957                    /lib/x86_64-linux-gnu/libreadline.so.6.2
7f155514b000-7f1555151000 rw-p 0003b000 fd:02 9577957                    /lib/x86_64-linux-gnu/libreadline.so.6.2
7f1555151000-7f1555152000 rw-p 00000000 00:00 0 
7f1555158000-7f155515f000 r-xp 00000000 fd:02 9577913                    /lib/x86_64-linux-gnu/librt-2.15.so
7f155515f000-7f155535e000 ---p 00007000 fd:02 9577913                    /lib/x86_64-linux-gnu/librt-2.15.so
7f155535e000-7f155535f000 r--p 00006000 fd:02 9577913                    /lib/x86_64-linux-gnu/librt-2.15.so
7f155535f000-7f1555360000 rw-p 00007000 fd:02 9577913                    /lib/x86_64-linux-gnu/librt-2.15.so
7f1555360000-7f1555378000 r-xp 00000000 fd:02 9577974                    /lib/x86_64-linux-gnu/libpthread-2.15.so
7f1555378000-7f1555577000 ---p 00018000 fd:02 9577974                    /lib/x86_64-linux-gnu/libpthread-2.15.so
7f1555577000-7f1555578000 r--p 00017000 fd:02 9577974                    /lib/x86_64-linux-gnu/libpthread-2.15.so
[... cut ...]
7f1555d46000-7f1555d47000 r--p 00016000 fd:02 17577104                   /usr/local/lib/libYARP_math.so.2.3.63.7
7f1555d47000-7f1555d48000 rw-p 00017000 fd:02 17577104                   /usr/local/lib/libYARP_math.so.2.3.63.7
7f1555d48000-7f1555d4a000 r-xp 00000000 fd:02 17577113                   /usr/local/lib/libYARP_init.so.2.3.63.7/home/travis/build.sh: line 41: 16218 Aborted                 testrunner --verbose --suit ../../icub-tests/suits/basics-icubGazeboSim.xml
The command "testrunner --verbose --suit ../../icub-tests/suits/basics-icubGazeboSim.xml" exited with 134.
~~~

apaikan, 2015-05-15:
@traversaro  Can you test again after updating yarp to HEAD? 
I  guess the error might be related to the issue #27!

traversaro, 2015-05-16:
Yes, [Travis is happy now!](https://travis-ci.org/robotology/gazebo-yarp-plugins/builds/62810324) (And RTF is now used in a CI system!)

closed 2015-05-16 by traversaro