robotology/robot-testing-framework #27: [yarp] If the application name is longer then 15 characters, the fixture does not load

traversaro, 2015-05-14:
If I tried to load with the yarpmanager fixture plugin an application with a name (contained in the xml `<name>` tag) longer or equal to 16 characters, I got a failure message: 
~~~bash
pegua@pareto:~/src/icub-tests/build$ testrunner --suit ../suits/basics-icubGazeboSim.xml --verbose
Loading ../suits/basics-icubGazeboSim.xml
Staring test runner.
Test suit Basic Tests Suite started...
[INFO]  (Basic Tests Suite) reports: yarpmanager is setuping the fixture...
[ERROR] (Basic Tests Suite) asserts error on (ret) with exception: yarpmanager cannot setup the fixture because Application is not loaded.
[INFO]  (Basic Tests Suite) reports: yarpmanager is tearing down the fixture...
Test suit Basic Tests Suite failed!
Ending test runner.
~~~

apaikan, 2015-05-15:
@traversaro Cannot reproduce the issue on Debian Wheezy!!! 

```xml
<suit name="Basic Tests Suite">
    <description>Testing robot's basic features</description>
    <environment>--robotname icubSim</environment>
    <fixture param="--fixture icubsim-fixture_loooooooooooooooooooooooooooooooooooooooo.xml"> yarpmanager </fixture>
...
```

the test run : 

```
$ testrunner -v -s ../suits/basics-icubSim.xml 
Loading ../suits/basics-icubSim.xml
Staring test runner.
Test suit Basic Tests Suite started...
[INFO]  (Basic Tests Suite) reports: yarpmanager is setuping the fixture...
Test case CameraTest started...
||| did not find right_camera.ini
||| did not find right_camera.ini
[INFO]  (Test Right Camera) reports: connecting from /CameraTest/image:i to /icubSim/cam/right
[INFO]  (Test Right Camera) reports: Reading images...
...
```

traversaro, 2015-05-15:
Sorry, I did not explain properly.

This fixture is working fine (`iCub Simulator` is 14 character) : 
~~~
<application>
    <name>iCub Simulator</name>
    <description>A fixture to prepare iCub Simulator for the test cases</description>
    <version>1.0</version>
    <authors>
        <author email="ali.paikan@iit.it">Ali Paikan</author>
    </authors>
    <module>
        <name>iCub_SIM</name>
        <parameters></parameters>
        <node>localhost</node>
    </module>    
    <module>
        <name>yarpview</name>
        <parameters>--name /right</parameters>
        <node>localhost</node>
        <prefix></prefix>
        <dependency>
            <port timeout="10">/icubSim/cam/right</port>
            <port timeout="10">/icubSim/cam/left</port>
        </dependency>
    </module>

    <connection>
        <from>/icubSim/cam/right</from>
        <to>/right</to>
    </connection>
</application>
~~~

while this is failing (`iCub Simulator Loooooooooooooooooooooong` is >> 16 characters):
~~~ 
<application>
    <name>iCub Simulator Loooooooooooooooooooooong</name>
    <description>A fixture to prepare iCub Simulator for the test cases</description>
    <version>1.0</version>
    <authors>
        <author email="ali.paikan@iit.it">Ali Paikan</author>
    </authors>
    <module>
        <name>iCub_SIM</name>
        <parameters></parameters>
        <node>localhost</node>
    </module>    
    <module>
        <name>yarpview</name>
        <parameters>--name /right</parameters>
        <node>localhost</node>
        <prefix></prefix>
        <dependency>
            <port timeout="10">/icubSim/cam/right</port>
            <port timeout="10">/icubSim/cam/left</port>
        </dependency>
    </module>

    <connection>
        <from>/icubSim/cam/right</from>
        <to>/right</to>
    </connection>
</application>
~~~

apaikan, 2015-05-15:
Thanks @traversaro. 
Fixed in yarp commit robotology/yarp cdb890c47c24418d306a91bf5891ec0164b9bbfe !

closed 2015-05-15 by apaikan