robotology/robot-testing-framework #30: Generate Jenkins compatible xml files

traversaro, 2015-05-15:
By writing a class similar to `TextOutputter` but that dumps the test results in a format compatible with Jenkins.

traversaro, 2015-05-15:
Apparently Jenkins (and also Atlassian Bamboo [1], so @luca-fiorio is happy) are digesting XML files in the JUnit XML format. Apparently there is not clean definition of this format, but a few XSD files around the net show that is a simple format:
https://svn.jenkins-ci.org/trunk/hudson/dtkit/dtkit-format/dtkit-junit-model/src/main/resources/com/thalesgroup/dtkit/junit/model/xsd/junit-4.xsd

[1] : https://confluence.atlassian.com/display/BAMBOO/JUnit+parsing+in+Bamboo

traversaro, 2015-05-18:
More info: http://help.catchsoftware.com/display/ET/JUnit+Format .

apaikan, 2015-05-18:
@traversaro  have you seen this?  http://www.yolinux.com/TUTORIALS/CppUnit.html

There is an example of the Kenkins XML file and how it has been implemented in the CppUnit.

traversaro, 2015-05-19:
Cool, I  guess for us (given that we have to write an XMLOutputter anyway) it make sense to directly use the JUnit XML format.

apaikan, 2015-10-12:
@traversaro  Could you have time to take a look at JUnit XML formatter?  I am playing around with that these days because we need to integrate RTF with jenkins. 

Just want be sure that we are not doing the same thing in parallel.

traversaro, 2015-10-12:
I had a non-working prototype locally, I'll check quickly if it can be finished in the short term!

apaikan, 2015-10-12:
@traversaro  okay. I have taken a look inside the XML file and apparently it is not complicated. Ill prepare some basic stuffs quickly within this week. You can later continue working on that. :)

traversaro, 2015-10-13:
Yes, I checked and I did not actually implement anything, so please go ahead!

apaikan, 2015-10-19:
done in commit d56339fa341f668cdb384fe9f2cf8fe0c6661b21.

closed 2015-10-19 by apaikan