robotology/robot-testing-framework #31: fixtures: timing problem on test startup

randaz81, 2015-05-20:
It seems that tests may fail because when they start, fixture is not ready yet to execute the test.
Example: I start a motor test on the robot simulator. Even if the simulator is immediately launched (and control ports are responsive), it take approx 10 seconds to perform an initial "calibration". PositionMove() commands are not correctly executed until the parts reach their final home position.
Of course this issue could be addressed by improving the state-machine of the simulator, however I'm wondering if it worth implementing a parametric delay after the fixture startup for applications (e.g. third party modules) which do not provide a better handshaking mechanism...

randaz81, 2015-05-20:
duplicate:
https://github.com/robotology/icub-tests/issues/11

apaikan, 2015-05-23:
fixed in https://github.com/robotology/yarp/commit/becdbbee597281cd459fd9c2a9e804b7f2c77e62 by specifying a wait after launching the modules using the `yarpmanager`. 
For example you can launch the iCub_SIM using the fixture manager as follows: 
```xml
    <module>
        <name>iCub_SIM</name>
        <parameters></parameters>
        <node>localhost</node>
        <ensure>
            <wait>10</wait>
        </ensure>
    </module>    
```

closed 2015-05-23 by apaikan