robotology/robot-testing-framework #32: [yarpmanager] the <ensure> and <dependency> tags are considered only for the first module in the fixture

traversaro, 2015-10-11:
Example: if I write a fixture like this one:
~~~
<application>
    <name>iCub Simulator</name>
    <description>A fixture to prepare iCub Simulator for the test cases</description>
    <version>1.0</version>
    <authors>
        <author email="ali.paikan@iit.it">Ali Paikan</author>
    </authors>
    <module>
        <name>iCub_SIM</name>
        <parameters></parameters>
        <node>localhost</node>
        <ensure>
            <wait>1</wait>
        </ensure>
    </module>
    <module>
        <name>yarpview</name>
        <parameters></parameters>
        <node>localhost</node>
        <ensure>
            <wait>100</wait>
        </ensure>
        <dependencies>
          <port timeout="5.0">/dsaadsdada</port>
        </dependencies>
    </module>    
</application>
~~~
Nor the ensure tag neither the dependency tag are considered for `yarpview`, and `yarpview` is simply launched immediately and the fixture is considered loaded.

traversaro, 2015-10-11:
cc @apaikan any pointers to some relevant class from which to start debugging?

apaikan, 2015-10-12:
Thanks @traversaro

There was a tiny bug in `Manager::allRunning()` from libYARP_manager which was not returning correct value when one of the modules is not running.  

Anyway, it is fixed on Yarp now. 
 
```
Staring test runner.
Test suit Basic Camera Tests Suite started...
[INFO]  (Basic Camera Tests Suite) reports: yarpmanager is setuping the fixture...
waiting on yarpview for 5.00 s.
[ERROR] (Basic Camera Tests Suite) asserts error on (ret) with exception: yarpmanager cannot setup the fixture because Failed to run yarpview on localhost. (State: 5, parameter: )
[INFO]  (Basic Camera Tests Suite) reports: yarpmanager is tearing down the fixture...
Test suit Basic Camera Tests Suite failed!
Ending test runner.
```

closed 2015-10-12 by apaikan