robotology/robot-testing-framework #34: modify RTF_TEST_CHECK so that it always prints the string and report success/failure

lornat75, 2015-10-15:
E.g.:

RTF_TEST_CHECK(value, "checking value")

should be logged as:

checking value...OK

or 

checking value... NOT OK

At the moment the macro prints the string only if the tests fails.

lornat75, 2015-10-15:
@traversaro @randaz81 @valegagge

apaikan, 2015-10-19:
I doubt which message user should include in `RTF_TEST_CHECK`; i.e. reports the cause of failure or a comment on what has been checked. For example: 

```c++
bool ret = yarp::os::Network::connect("/port1", "/port2");
RTF_TEST_CHECK(ret, "connecting two ports.");    // what was the reason of failure?
```
or 

```c++
bool ret = yarp::os::Network::connect("/port1", "/port2");
RTF_TEST_CHECK(ret, "cannot connect two ports because blablabla..." ); 
```

for now this can be solved by using two different macros (maybe not very convenient): 

```c++
RTF_TEST_REPORT("connecting two ports.");
bool ret = yarp::os::Network::connect("/port1", "/port2");
RTF_TEST_CHECK(ret, "cannot connect two ports because blablabla..." ); 
```

The `RTF_TEST_CHECK` macro has been used in many different test cases and changing it may needs changes to some of the tests implementation :s What about introducing another macro which aggregate both (comment and cause of failure) ?

lornat75, 2015-10-19:
I vote for the first option because you rarely know why a test fails but you do known what you are testing.

RTF; i think we just need to extend the macro so that it always reports the result of the test (either success or fail).

apaikan, 2015-11-03:
* the `RTF_TEST_CHECK` has been modified to shows the corresponding messages (thanks to @lornat75 ) 

* for the backward compatibility, the `RTF_TEST_FAIL_IF`  has been implemented which does the same job as the old `RTF_TEST_CHECK`

* for example, the following code:
```c++
RTF_TEST_CHECK(2<3, "two is less than three");
```
will result: 
```
[INFO]  (MyTest) checks: two is less than three
```
* and the following code: 
```c++
RTF_TEST_CHECK(2==3, "two is equal to three");
```
will result: 
```
[FAIL]  (MyTest) checks: two is equal to three
```

closed 2015-11-09 by apaikan