robotology/robot-testing-framework #37: Python travis build failing

traversaro, 2015-12-16:
https://travis-ci.org/robotology/robot-testing/jobs/96977269

Actual error: 
~~~
[ 60%] Building CXX object src/plugins/python/CMakeFiles/RTF_python.dir/src/PythonPluginLoader.cpp.o
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp: In member function ‘RTF::TestCase* RTF::plugin::PythonPluginLoaderImpl::open(std::string)’:
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:112:49: error: ‘PyString_FromString’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:135:52: error: ‘Py_InitModule4’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp: In member function ‘std::string RTF::plugin::PythonPluginLoaderImpl::getPythonErrorString()’:
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:183:37: error: ‘PyString_Check’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:184:44: error: ‘PyString_AsString’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:191:37: error: ‘PyString_Check’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:192:60: error: ‘PyString_AsString’ was not declared in this scope
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp: In member function ‘virtual bool RTF::plugin::PythonPluginLoaderImpl::setup(int, char**)’:
/home/travis/build/robotology/robot-testing/src/plugins/python/src/PythonPluginLoader.cpp:231:52: error: ‘PyString_FromString’ was not declared in this scope
make[2]: *** [src/plugins/python/CMakeFiles/RTF_python.dir/src/PythonPluginLoader.cpp.o] Error 1
make[1]: *** [src/plugins/python/CMakeFiles/RTF_python.dir/all] Error 2
make: *** [all] Error 2
~~~

Googling for similar errors it appears that the bug is caused by a mismatch between the Python versions:
https://bugs.launchpad.net/opencog/+bug/1063039
http://sourceforge.net/p/robotcub/mailman/robotcub-hackers/thread/4AB3E428.5020601@alum.mit.edu/

Probably now the Travis images have been update to have multiple versions of Python, and there is a bug in the CMake code handling Python versions. I will look into it.

apaikan, 2015-12-21:
@traversaro  which versions of python have been installed? 

anyway, there is a cmake option `RTF_USE_PYTHON_VERSION` to force using specific version of python for binding. can you please check it?

traversaro, 2016-03-21:
Checked it, the problem is that some RTF is not compatible with Python 3. Working on it right now.

traversaro, 2016-03-21:
Relevant link: https://docs.python.org/2/howto/cporting.html .

traversaro, 2016-03-21:
Giving up as for the moment I don't know enough of Python internals to do it (especially to properly substituted the `Py_InitModule4` function), just preselecting Python 2 in the cmake for now and opening a new issue to track Python 3 support.

traversaro, 2018-06-08:
Relevant: https://pythonclock.org/ .

traversaro, 2019-02-01:
The Python clock is rapidly ticking. : ) 
If we do not think we can commit effort to fix this, probably the cleanest thing to do is just to drop Python support in RTF. 
@aerydna @drdanz @lornat75

apaikan, 2019-02-01:
Hi guys,
allways glad to hear from you even it is an issue ;)
I am a bit out of the discussion. we are using RTF mostly woth python (2.7) and everything is fine. is the main issue with Python 3.x?

traversaro, 2019-02-01:
Hi @apaikan , 
    happy to hear from you, hope everything is ok.

Back in the days I tried to get RTF to work on Python 3, but I was unable to port all the part using `Py_InitModule4` (around  https://github.com/robotology/robot-testing/blob/c7c00459f721648c8f74f973a881d9078a206c50/src/plugins/python/src/PythonPluginLoader.cpp#L134) to use Python 3, that changed that APIs. https://docs.python.org/3/howto/cporting.html is probably useful,  but I was unable to get it to work. 

Given that Python 2 will be officially unsupported in less than 10 months (see https://pythonclock.org/) I was trying to raise awareness on this issue.

aerydna, 2019-02-09:
i am for dropping the compatibility

drdanz, 2019-10-18:
2 months left to python 2 end of life, and still no python 3 support :disappointed:

Nicogene, 2019-10-21:
But what is the main incompatibility issue?
What it is blocking ?

traversaro, 2019-10-21:
The C API of the Python interpreter changed between Python 2 and Python 3, and we use it in https://github.com/robotology/robot-testing-framework/blob/c7c00459f721648c8f74f973a881d9078a206c50/src/plugins/python/src/PythonPluginLoader.cpp#L134 . Some notes about a previous failed attempt to port (but I did not spent a lot of time on it) can be found in https://github.com/robotology/robot-testing-framework/issues/37#issuecomment-459850630 .

drdanz, 2020-01-09:
Python 2 reached its end of life