robotology/robot-testing-framework #46: Fixture incorrectly consider launched even if one of its modules was not launched

traversaro, 2016-03-25:
If one of the modules of a fixture is not launched because its `<dependencies>` are not satisfied, yarpmanager consider the fixture to be successful anyway, launching the tests.

apaikan, 2016-12-05:
I have tried to reproduce the bug but i could not :( 

In my setup, i have a fixture with two module in which the second one (i.e.`yarpview`) fails on waiting for a dependency (i.e. a port). As it's expected, the `testrunner` dos not run the tests! 

@traversaro  Can you provide an example?  
   
- my fixture file: 
```xml
<application>
    <name>iCub Simulator</name>
    <description>...</description>
    <module>
        <name>test_grabber</name>
        <parameters>--name /grabber</parameters>
        <node>localhost</node>
        <deployer>yarpdev</deployer>
    </module>    

    <module>
        <name>yarpview</name>
        <parameters>--name /view</parameters>
        <node>localhost</node>
        <dependencies>
            <port>/NOT/EXISTING/PORT</port>
        </dependencies>
    </module>
</application>
```

- my testrunner : `testrunner -v -s suits/basics-icubSim.xml`
```bash
Loading suits/basics-icubSim.xml
Staring test runner.
Test suit Basic Tests Suite started...
[INFO]  (Basic Tests Suite) reports: yarpmanager is setuping the fixture...
[ERROR] (Basic Tests Suite) asserts error on (ret) with exception: yarpmanager (run) cannot setup the
fixture because Failed to run yarpview on localhost. (State: 5, parameter: --name /view) 
cannot run yarpview on localhost : "Timeout while waiting for some resources. "
[INFO]  (Basic Tests Suite) reports: yarpmanager is tearing down the fixture...
Test suit Basic Tests Suite failed!
Ending test runner.

---------- results -----------
Total number of test suites  : 1
Number of passed test suites : 0
Number of failed test suites : 1
Total number of test cases   : 0
Number of passed test cases  : 0
Number of failed test cases  : 0
```

traversaro, 2016-12-05:
I do not remember anything about this, it is also possible that it was fixed by some fix on YARP side. : ) 
I think it is safe to assume that the bug is fixed, and we can reopen if we find again this problem.

apaikan, 2016-12-05:
@traversaro  I just noticed that this issue is very old :grimacing:

closed 2016-12-05 by apaikan