robotology/robot-testing-framework #47: Confusion between 'suit' and 'suite'

gsaponaro, 2016-11-23:
Hi @apaikan , thanks for providing this tool.

I'm just curious why in some cases the expression "suit" is used ([here](https://github.com/robotology/robot-testing/search?utf8=%E2%9C%93&q=suit)), whereas other times it's "suite" ([here](https://github.com/robotology/robot-testing/search?utf8=%E2%9C%93&q=suite)).

If it was just a typo, perhaps it's worth changing it to be consistent everywhere?

Feel free to close if you consider this a minor/irrelevant issue. :)

apaikan, 2016-11-23:
:laughing:  

There was an initial typo (`suit` in place of `suite`), also in the class name, ` RTF::TestSuit`! 
Would be happy to see they are corrected but unfortunately they are many . 

let change everything to **sweet**  ;)

gsaponaro, 2017-10-02:
Fixed by https://github.com/robotology/robot-testing/pull/71

closed 2017-10-02 by gsaponaro