robotology/robot-testing-framework #48: CMake warnings under macOS

claudiofantacci, 2016-12-04:
While **configuring** CMake, I get the following warning:
```
CMake Warning (dev) at src/rtf/CMakeLists.txt:8 (project):
   Policy CMP0048 is not set: project() command manages VERSION variables.
   Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
   command to set the policy and suppress this warning.

   The following variable(s) would be set to empty:

     RTF_VERSION
     RTF_VERSION_MAJOR
     RTF_VERSION_MINOR
     RTF_VERSION_PATCH
     RTF_VERSION_TWEAK
 This warning is for project developers.  Use -Wno-dev to suppress it.
```

During **generation** I get:
``` 
Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
   set the policy and suppress this warning.

   MACOSX_RPATH is not specified for the following targets:

    RTF
    RTF_dll

 This warning is for project developers.  Use -Wno-dev to suppress it.
```

System: macOS 10.11.6 (El Capitan), CMake 3.7.0.

claudiofantacci, 2016-12-19:
Project version shall be set in the `project()` command as reported [here](https://cmake.org/cmake/help/v3.0/command/project.html).

In the CMakeLists.txt the `include(RTFVersion)` should export a variable to be used in `project()', e.g.:
```
inlcude(RTFVersion)
project(RTF LANGUAGE CXX VERSION ${RTFVersionNumber})
...
```
This should be compliant with [CMP0048](https://cmake.org/cmake/help/v3.0/policy/CMP0048.html#policy:CMP0048)
cc: @apaikan @traversaro @drdanz

traversaro, 2016-12-19:
I am quite confused. If we use the new signature of `project` we break compatibility with CMake < 3.0 (not a big problem, considering https://github.com/robotology/QA/issues/173 and given that `RTF` is mostly a developer tool). However I wonder why we don't get this warning in the other libraries.

For the RPATH warning, the logic to set it properly can be found in https://github.com/robotology/ycm/blob/master/modules/AddInstallRPATHSupport.cmake .

claudiofantacci, 2016-12-19:
As far as VERSION issue is concerned, @apaikan is also experiencing this warning on Windows. What I would guess about it, is that with the latest version of CMake (I have 3.7.1), developers are pushing to move users toward 3.0>. I have no problem with supporting older version (say 2.8>), but we may want to consider enforcing the policies and moving toward CMake 3.0.

As far as RPATH issue is concerned, I think we can use [AddInstallRPATHSupport.cmake](https://github.com/robotology/ycm/blob/master/modules/AddInstallRPATHSupport.cmake) directly copying it into the `config` folder (and modifying CMakeLists accordingly). Am I right?

drdanz, 2016-12-20:
I think this is the problem:

[CMakeLists.txt:9](https://github.com/robotology/robot-testing/blob/master/CMakeLists.txt#L9)
```cmake
project(RTF)

# ...

# include version number
include(RTFVersion)
```

[RTFVersion.cmake](https://github.com/robotology/robot-testing/blob/master/conf/RTFVersion.cmake)
```cmake
set(RTF_VERSION_MAJOR "1")
set(RTF_VERSION_MINOR "0")
set(RTF_VERSION_PATCH "1")
set(RTF_VERSION_TWEAK "0")

# Generate RTF_VERSION
if(RTF_VERSION_TWEAK)
    set(RTF_VERSION "${RTF_VERSION_MAJOR}.${RTF_VERSION_MINOR}.${RTF_VERSION_PATCH}.${RTF_VERSION_TWEAK}")
else()
    set(RTF_VERSION "${RTF_VERSION_MAJOR}.${RTF_VERSION_MINOR}.${RTF_VERSION_PATCH}")
endif()
```

[robot-testing/src/rtf/CMakeLists.txt:8](https://github.com/robotology/robot-testing/blob/master/src/rtf/CMakeLists.txt#L8)
```cmake
project(RTF)
```

The second `project(RTF)` is overriding the values set by `RTFVersion`. CMake checks if the variables set are empty and warns the uses. Just removing the second call (that is actually useless and confusing since it will overwrite several variables) should fix this warning.
@claudiofantacci Can you please try?

claudiofantacci, 2016-12-20:
I'm on it 👍

commit bab1b6be (2016-12-29): Fix project() VERSION issue on whole project

To avoid generating VERSION policy issue CMP0048 while
configuring the project, the project root CMakeLists and the RTF
CMakeLists have been modified.
In particular, in RTF CMakeLists.txt the project(RTF) call has been
removed because it was redundant and, as a conseguence, the cause of the
version policy issue.
As a conseguence, several variable in RTF CMakeLists.txt have been
modified to account for the new path of the RTF project which is
located in src/rtf/* folder and not in the root one where
project(RTF) is originally defined.

Fixes #48
closed 2016-12-29 by apaikan