robotology/robot-testing-framework #54: Having multiple fixture manager for each suite

apaikan, 2016-12-29:
It will be useful (and in some cases necessary) to have multiple fixture manager for each suites. Example: 

```xml
<suit name="Basic Tests Suite">
    <description>Testing basic features</description>
    <fixture param="--param myparam"> fixturemanager1 </fixture>
    <fixture param="--param myparam"> fixturemanager2 </fixture>
    ...

    <!-- tests -->
    ...
</suit>

```

closed 2017-01-05 by apaikan

pattacini, 2017-01-05:
Not being this available yet, I struggled to make a single fixture work with all inside. However, this gave me the chance to fix a subtle bug while closing the `cartesiancontrollerserver` component.

Once more let me point out that **RTF** is a magic tool 🥇
I've been using it to create the backbone infrastructure that enables **automatic grading**, in view of VVV17 😄

pattacini, 2017-01-05:
Well, even though we have multiple fixtures, is there a closing order that we can ensure? Otherwise, we fall back again in the same problem, I'm afraid.