robotology/robot-testing-framework #55: moving yarp middleware plugin to yarp repository

apaikan, 2016-12-29:
For now, yarp middleware plugin in RTF depends on yarp.  This means that that it will be very complicated to use RTF to develop units tests within yarp simply because of mutual dependency!

**1)** The solution suggested by @drdanz is to move the RTF yarp middleware plugin to the yarp repository. In this way one can: 
- Install RTF on a machine 
- Install yarp with RTF enabled (can be automatically done by cmake) 
- compile and run yarp test units which are developed using RTF (e.g.  `make tests`) 

**2)** The second option is to have the yarp unit tests in a separate repository. This needs a bit of more work to compile and run the tests especially if we want to run the new tests (i.e. written using RTF) along with the available tests in YARP.

lornat75, 2017-01-03:
I think it is a good solution. I don't see problems with it.

apaikan, 2017-01-09:
@valegagge has a Yarp middleware specific utility library within RTF. At my first check I do not see any problem with moving them to YARP. 

@valegagge would you please confirm?

drdanz, 2017-01-09:
The only issue I foresee is that at the moment, in order to ensure that the library is available, you have to run something like:

```cmake
find_package(RTF COMPONENTS YARP REQUIRED) # Checks if `RTF` is available, and if it was built with the `YARP` component.
```

If we want to keep using this, we need to do some sort of hack in RTFConfig.cmake. I think it is doable, but it will require some more effort, and I don't think it is worth.

The alternative is to do the opposite, i.e. something like:

```cmake
find_package(RTF REQUIRED) # Checks if `RTF` is available
find_package(YARP COMPONENTS RTF REQUIRED) # Checks if `YARP` is available, and if it was built with the `RTF` component.
```

Does this look ok to you?

valegagge, 2017-01-10:
@apaikan : yes!

randaz81, 2017-01-10:
sorry guys, I'm a bit lost... 😳 
what is the 
> yarp middleware plugin in RTF depends on yarp

I can't find a yarp plugin inside `robotology\robot-testing\src\plugins`.
Do you mean `robotology\robot-testing\src\middleware\yarp`?  
Btw what is `robotology\yarp-tests` @barbalberto ?
And what is rule to follow when you have to choose if a test has to be put a icub-tests, yarp-tests or in harness_xxx?
Thank you!

barbalberto, 2017-01-11:
The mentioned plugin is the hook that RTF uses to work with YARP. RTF is by itself middleware independent so you can use it with yarp, ros or whatever, as long as a plugin is provided for the specified middleware.

Right now this plugin is inside RTF repo (I don't remember where), and for this reason we cannot write tests inside yarp harness_xxx using RTF, because yarp will depend on RTF which, in turns, depends on yarp.
The idea is to move this plugin into YARP repo in order to fix this circular dependency issue.

About which test shall be in harness and which shall use RTF, I share my personal idea:
- harness: basic unit test that uses only modules inside yarp, regression tests, basic functionality like
  - test of ports, carriers
  - wrapper / client test
  - RF module
  - lib math test
  - basically those tests are looking for bugs in the code and shall test small portion of code at a time, (remember that harness tests are always run after each push on robotology)
- RTF: more complex tests which falls into one or more of the following cases:
  - a big number of modules needs to be run
  - long execution time (this will slow down jenkins too much in continuous integration)
  - uses the simulator
  - non YARP modules are required to do the test

In my personal account I started a repo because I'm gonna write some tests for the clock and I need a lot of modules to be started (a clock producer, different consumers) and I want to test it also with ROS so I need RTF. I started there because, for the plugin reason, I could not start RTF tests in yarp.
When the plugin is succesfully integrated, I'll move the tests into yarp repo.

Hope it helps

traversaro, 2017-06-30:
Fixed in https://github.com/robotology/yarp/pull/1097 .

closed 2017-06-30 by traversaro