robotology/robot-testing-framework #57: CTest integration v2

traversaro, 2017-01-11:
To simplify adding tests using `testrunner` to CTest using the `add_test` cmake command, the CMake infrastructure of RTF should provide a variable containing the absolute path to the testrunner. 

Helper CMake functions to automatically add a CTest test from a suite would also be convenient.

cc @S-Dafarra

traversaro, 2017-01-13:
Use case : https://github.com/S-Dafarra/Operational_Control_with_iCub/pull/1/files#diff-15547c54d3d4898a882b4ab7b3cee381R29 .

apaikan, 2017-01-15:
I have added some simple tests which can be run using `testrunner` via `make test` (see `testing` branch) .
@traversaro and @drdanz  we can elaborate that!

S-Dafarra, 2017-01-15:
I had an issue using ``make test``: the simulator window appeared and disappeared immediately, making the test fail due to fixtures problems. This issue didn't come out using ``ctest`` though.

traversaro, 2017-09-27:
cc @aerydna @drdanz

traversaro, 2017-10-30:
Fixed by https://github.com/robotology/robot-testing/pull/88 .

apaikan, 2017-11-14:
@traversaro  is this issue fixed? :p

drdanz, 2017-11-20:
We now have a `rtf_add_suite` CMake command, perhaps also an `rtf_add_testcase` could be useful? Otherwise I think this could be closed.

traversaro, 2017-11-20:
I think the `rtf_add_suite` is enough unless we actually need a `rtf_add_testcase` in a real-world scenario.

closed 2017-11-24 by aerydna