robotology/robot-testing-framework #64: Doxygen generated documentation is outdated

traversaro, 2017-06-27:
The doxygen generated documentation available at http://robotology.github.io/robot-testing/index.html is currently outdated (it contains the documentation of the 1.0 version and it was generate the last time the 5th of January).

traversaro, 2017-06-27:
cc @Nicogene

traversaro, 2017-06-29:
For the time being, I added a script for updating the doxygen documentation and uploading it automatically to [Gitlab Pages](https://about.gitlab.com/features/pages/) in the https://gitlab.com/robotology/docs . It is a bit of an experiment, but it is working fine:
* `master` branch: http://robotology.gitlab.io/docs/rtf/master/
* `devel` branch: http://robotology.gitlab.io/docs/rtf/devel/

drdanz, 2017-06-29:
@traversaro cool, I didn't know we had a group on gitlab as well :)

Nicogene, 2017-06-29:
I updated the documentation in `icub-tests`, but I think we should add a section in yarp documentation about rtf plugins (if it is not already present) and then add references to it in `icub-tests` and `rtf` documentation. 
What do you think about it?

pattacini, 2017-06-29:
It should be sufficient to use the local `gh-pages` branch to generate the documentation, or even better we could put in place the automatic mechanism that relies on `travis CI` to produce the doxygen and push to GitHub.

Thanks to @francesco-romano, we have instructions for this available here: `how-to-document-modules/auto_doc`.

[1]: https://github.com/robotology/how-to-document-modules/tree/master/auto_doc

alecive, 2017-06-29:
> It should be sufficient to use the local gh-pages branch to generate the documentation, or even better we could put in place the automatic mechanism that relies on travis CI to produce the doxygen and push to GitHub.

..with cron jobs that automatically re-generate the doxy every week (or every day) even if nobody committed to the repo :grin:

FYI I love `Travis CI`'s cron jobs!

traversaro, 2017-06-30:
+1 for documenting YARP rtf plugins in YARP docs.

aerydna, 2017-10-03:
solved by #73

closed 2017-10-03 by aerydna