robotology/robot-testing-framework #67: RTF_TEST_FAIL_IF macro logic is inverted

aerydna, 2017-08-17:
the RTF_TEST_FAIL_IF macro fail if the condition is false. example:
int a = 5;
int b = 2;
RTF_TEST_FAIL_IF(a > b) //this call does not fail as it should..

traversaro, 2017-08-18:
Not sure, but perhaps that macro was supposed to just be a backward compatibility layer and meant to be deprecated: https://github.com/robotology/robot-testing/issues/34 .

pattacini, 2017-08-18:
If I'm not mistaken, its use is quite spread at least in vvv-school. Unable to check right now.

pattacini, 2017-08-23:
Apparently, I was wrong saying the macro is used in [`vvv-school`](https://github.com/vvv-school). I've just checked and it's never used, instead.

However, [`icub-tests`](https://github.com/robotology/icub-tests) make use of it quite extensively. To this end, try out [this search][1].

[1]: https://github.com/search?p=3&q=org%3Arobotology+RTF_TEST_FAIL_IF&type=Code

apaikan, 2017-09-06:
@aerydna : Please pay attention that "RTF_TEST_FAIL_IF" fails if the condition **is not hold**! This standard has been chosen according to CppUnit testing framework! 
According to its documentation: 
```
/** Conditional failure report. RTF_TEST_FAIL_IF does not throw any
 *  exception. It reports a failure message to the TestResult indiecating
 *  that the check on the corresponding condition evaluates to \c false.
 * \ingroup Test Assertions
 * \param message Message to be reported as the detail of TestMessage
 * \param condition If this condition evaluates to \c false then the
 *                  test failed.
 *
 * \note RTF_TEST_FAIL_IF throws error exception if it is not called
 *       within a TestCase class.
 */
```

aerydna, 2017-09-06:
@apaikan  is there a CppUnit equivalent of `RTF_TEST_FAIL_IF`? the signature is the same? because as it is (with a `IF`) i tend to give the macro the following meaning:

```
RTF_TEST_FAIL_IF(condition)

==

if (condition)
{
    fail();
}
```

apaikan, 2017-09-07:
@aerydna: Agree with you!  I guess my confusion came from the implementation of `CPPUNIT_ASSERT` macro:  

```cpp
#define CPPUNIT_ASSERT ( condition ) 	
( CPPUNIT_NS::Asserter::failIf( !(condition),                                    \
                                 CPPUNIT_NS::Message( "assertion failed",         \
                                                      "Expression: " #condition),  \
                                 CPPUNIT_SOURCELINE() ) )
```

Please go ahead to change the the condition inside. If you are eager to do that, please remember to

- Update the documentation (doxygen)
- Check if there is any example regarding that which need to be updated 
- and **more importantly**, where it has been used in the **bindings** (Lua, Python, Ruby, Ada,..)

pattacini, 2017-10-06:
@aerydna please remember to notify me when such changes happen, 'cause #84 is not only related to `RTF_TEST_FAIL_IF` as discussed here but also to e.g. `RTF_ASSERT_ERROR_IF` for which, instead, we'll have an impact on `vvv-school` code I'm maintaining. 

Honestly, I don't like to discover this by myself: I could easily miss reading PR for example 😉

pattacini, 2017-10-06:
More on this: [`icub-tests`][1] needs to be updated now.

@aerydna is this action already planned?

[1]: https://github.com/robotology/icub-tests

closed 2017-10-08 by pattacini