robotology/robot-testing-framework #68: order and dependencies in TestSuit

aerydna, 2017-09-26:
According to @traversaro it would be important to control the order of execution of both tests and fixtures.
while usual unit testing have to be order independent, in a robot testing scenario it would be useful to test the small component at first (like encoder response, plugin response etc) and then larger behaviour like for instance navigation and grasping.
for fixture it is necessary where some fixture (speaking of the yarp framework a yarp device, yarp application or the yarpmanager itself for example) need other utility (like yarp server) in order to start correctly.

as far as i know this functionality is not avaiable in rtf. the ways to implement it are the following

in my opinion the simplest and most intuitive solution from both the user and the development side is to maintain the order of the xml tags (top-bottom). this can be done by simply change the tests and fixtures data storage for from std::set to std vector. While the loss in term of performance is negligible in my oprinion an inconsistency with the xml parsing policy of yarp (where only the depth order of the tags is relevant). However being RTF and YARP two completely separate frameworks this shouldn't be a problem.

TLDR VERSION HERE:
There is no way to control the fixture or test execution order in RTF while in some cases this is a requirement. lets discuss on how implement it.

traversaro, 2017-09-26:
cc @drdanz @randaz81 @pattacini

traversaro, 2017-09-27:
Relevant discussion on stack overflow: https://stackoverflow.com/questions/1131495/in-xml-is-order-important . 
In general I find it makes sense to have different behavior between yarpmanager and rtf because `yarpmanager` launches external processes (whose order of execution is arbitrary by definition, depending on the OS scheduler) while RTF fixture and tests plugins are classes whose methods are launched in a sequential order by the `testrunner` process, and hence it make sense to define a sequence for them.

aerydna, 2017-10-02:
fixed by #69

closed 2017-10-02 by aerydna