robotology/robot-testing-framework #101: Rename RTF

drdanz, 2019-03-09:
* The acronym `RTF` is quite popular, the most accepted is `Rich Text Format`, but there are [many others](https://www.acronymfinder.com/RTF.html) meanings.
* There is a [librtf](http://librtf.sourceforge.net/)

This causes conflicts with other packages, making it impossible to package it properly.

I suggest to rename everything to something else.

* The libraries (`libRTF`, `libRTF_dll`, etc)
* The namespace (`RTF`)
* The include path (`include/rtf`)
* The CMake name (`RTF`)
* The CMake targets and components (`RTF`, `RTF_dll`, ecc)
* Exported CMake variables (`RTF_LIBRARIES`, `RTF_INCLUDE_DIRS`, ecc)
* The `testrunner` executable (`testrunner`)

This is obviously a disruptive change, therefore we need to discuss this, but I believe such a change is required, and we should do this change as soon as possible

Proposal for the names? I was thinking `RoboTF`, but TF in robotics has a different meaning. Maybe `RoboTest`?

pattacini, 2019-03-09:
What are the conflicts you discovered?

drdanz, 2019-03-10:
There are too many `librtf-xxx` packages related to rtf files, and the `libRTF.so` file requires a `librtf1` package. Moreover the headers are installed in `/usr/include/rtf`, it's quite confusing

traversaro, 2019-03-10:
I think expanding all the conflicting uses of the RTF acronym (library robot-testing-framework, headers prefix RobotTestingFramework, cmake package RobotTestingFramework, etc etc) is probably the best option

drdanz, 2019-03-11:
Perhaps when we do the change we should release it as RobotTestingFramework 2.0?

traversaro, 2019-03-11:
Good idea.

drdanz, 2019-03-14:
Actually I'm thinking that `RobotTestingFramework` is a very long name, not good for namespaces, cmake variables, etc. What about `RoboTest`?

traversaro, 2019-03-14:
I personally would prefer to minimize the changes, and I do not think that `RobotTestingFramework` is too long as a name. By the way, we can also think of just changing the things that are risk more conflicts, such as the CMake package name and the library name, and just keep RTF for anything else.

pattacini, 2019-03-14:
If we agree to proceed as @drdanz suggested, then I would go for @traversaro's proposal.

cc @lornat75

drdanz, 2019-03-15:
Ok, if you agree, I'd proceed with:

* Libraries and executables:
```
librobot-testing-framework.so
librobot-testing-framework-dll.so
robot-testing-framework-testrunner
```

* CMake package, targets and variables:
``` cmake
find_package(RobotTestingFramework)
target_link_libraries(foo RobotTestingFramework::RTF RobotTestingFramework::RTF_dll)
message("${RobotTestingFramework_INCLUDE_DIRS}")
message("${RobotTestingFramework_LIBRARIES}")
```

* Include path and namespace
```c++
#include <robot-testing-framework/TestCase.h>
RobotTestingFramework::TestCase
```
this can eventually be simplified during the port by using a namespace alias
```c++
#include <robot-testing-framework/TestCase.h>
namespace RTF = RobotTestingFramework;
RTF::TestCase
```

This is a big breaking change, therefore, if we go with this, we should decide how to proceed with the other repos. I'd like to merge this in master, and release 2.0 ASAP, but then all the other master repo would be broken. Perhaps we should do it together with YARP and iCub releases?

traversaro, 2019-03-15:
I never saw the hyphen used in the include director, it is ok to use it?

drdanz, 2019-03-15:
good question... I suppose it is, `ls -d /usr/include/*-*` has several results, even though most of them are just for separating the name and the version.
I don't want to risk though, let's just remove all the `-` (at this point also from the libs)
I'd make also the namespace lowercase, so that it matches with the include dir

* Libraries and executables:
```
librobottestingframework.so
librobottestingframework-dll.so
robottestingframework-testrunner
```

* CMake package, targets and variables:
``` cmake
find_package(RobotTestingFramework)
target_link_libraries(foo RobotTestingFramework::RTF RobotTestingFramework::RTF_dll)
message("${RobotTestingFramework_INCLUDE_DIRS}")
message("${RobotTestingFramework_LIBRARIES}")
```

* Include path and namespace
```c++
#include <robottestingframework/TestCase.h>
robottestingframework::TestCase
```
this can eventually be simplified during the port by using a namespace alias
```c++
#include <robottestingframework/TestCase.h>
namespace RTF = robottestingframework;
RTF::TestCase
```

drdanz, 2019-03-15:
BTW, I think that for coherence we should rename this repo `robotology/robottestingframework` as well, what do you think

traversaro, 2019-03-15:
`robot-testing-framework` is probably more consistent with the rest of the repository's names.

drdanz, 2019-04-10:
Done in #103

commit 265a513c (2019-06-03): Use Robot Testing Framework instead of RTF everywhere

The RTF acronym is usually used for Rich Text Format, and it is causing
conflicts in packaging (Fixes #101).

Big CMake refactor, update modules from YCM, use targets instead of
internal properties, move CMake files from 'conf' to 'cmake'.

Cleanup c++ files.

Update and fix copyright headers.

Bump version patch.

Change COMPATIBILITY to SameMajorVersion
closed 2019-04-10 by drdanz