robotology/robot-testing-framework #109: Permit to specify directory where to load plugins from without using LD_LIBRARY_PATH/PATH

traversaro, 2020-04-26:
See discussion https://github.com/robotology/robotology-superbuild/issues/95 for more info.

drdanz, 2020-04-27:
Related: robotology-playground/sharedlibpp#7

traversaro, 2024-11-08:
For `qpsolvers-eigen` I implemented yet another improvement, that I think is quite convenient and permits us to avoid setting any env variables, at least in most cases. 

The main library is called `<prefix>qpsolvers-eigen<extension>` (where `<prefix>`  and `<extension>` are os-specific values), while plugins are called in `<prefix>qpsolvers-eigen-osqp<extension>`, ``<prefix>qpsolvers-eigen-proxqp<extension>` etc etc. Differently from rtf that installs the main library in `<prefix>/lib` (or `<prefix>/bin` for Windows) and the plugins in  `<prefix>/lib/rtf`, for qpsolvers all libraries are installed in `<prefix>/lib`, as anyhow they are not ambiguous thanks to the `qpsolvers-eigen-` prefix in their name. 

Then, in the main library I extend the search path based on the location of the main  `<prefix>qpsolvers-eigen<extension>` library (see https://github.com/ami-iit/qpsolvers-eigen/blob/dfd9b3e326e27f8f73646070d120c53e42c4fbda/core/QpSolversEigen/Solver.cpp#L102-L110 and https://github.com/ami-iit/qpsolvers-eigen/blob/dfd9b3e326e27f8f73646070d120c53e42c4fbda/core/QpSolversEigen/Solver.cpp#L21-L57), so that as long as the library and the plugins are installed in the same install prefix (as it happens with conda packages, robotology-superbuild and colcon workspace with `--merge-install` option) the plugins can be found without the need of setting any env variable. 

In case plugins are installed in a separate install prefix with respect to the main `<prefix>qpsolvers-eigen<extension>` library, then the install directory can be added to the `SHLIBPP_PLUGIN_PATH` env variable (or a custom env variable as done by `blockfactory` with `BLOCKFACTORY_PLUGIN_PATH`, see https://github.com/robotology/blockfactory/blob/9921b59725b31cc5a655b9bdb316c634df6085d8/sources/Core/src/FactorySingleton.cpp#L30 .